prusa3d/PrusaSlicer #12491: 2.7.3 alpha vase degradation

IgorOhrimenko, 2024-03-21:
### Description of the bug

2.7.2 [image]
2.7.3
[image]
 [image]

### Project file & How to reproduce

Slice gap closing raduis: 0
[image]

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.3 alpha 1

### Operating system

Win10

### Printer model

Qidi X-Plus II

vovodroid, 2024-03-22:
Do you use absolute or relative E distance?

IgorOhrimenko, 2024-03-22:
@vovodroid I create the simple cone, open it at 2.7.2 and 2.7.3 and save project with all settings:
[cone-2.7.2.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14728752/cone-2.7.2.3mf.zip)
[cone-2.7.3.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14728753/cone-2.7.3.3mf.zip)

neophyl, 2024-03-22:
I see what you mean Vovodroid.  Changing to relative extrusion and adding in the G92 E0 to the after layer change field allows it to slice properly.

[image]

So is vase mode broken for absolute extrusion setups ?

IgorOhrimenko, 2024-03-23:
`relative extrusion and adding in the G92 E0` works, but it wasn't necessary before.

kubispe1, 2024-03-25:
I think we have fixed this issue already. I checked it with **PrusaSlicer-2.7.3-alpha1+6-win64**
Should be fixed with next release.
SPE-2198
[image]
Thanks

closed 2024-03-25 by kubispe1

l29ah, 2025-01-15:
fixed in 6c6f90fe4b9a059adde93b7cd32c8b2ba98e555a