prusa3d/PrusaSlicer #12492: Auto Calculate Purge Volume

TripleOneAustin, 2024-03-21:
Would be cool to have a auto calculate purge volume like in Bambu Slicer. This would just give a rough starting point for the purge volumes that you could then modify.

NotSwedishBacon, 2024-06-02:
I was just about to post something like this. I've been copying values over from orcaslicer at 0.4 multiplier and getting great results. It's annoying to have two slicers open for one feature so I'm looking forward to this being a native feature

tommylux, 2025-01-20:
+1 Have just discovered this feature in Orcaslicer, Would love to see auto calculate purge volume based on colour and multipler.

n0dr4c, 2025-03-19:
+1 please this feature is really a must have one! I am upset to launch orcaslicer to determine how many purge volume is necessary from one color to another and copy paste datas...

closed 2026-09-29 by prusaslicer-bot[bot]