prusa3d/PrusaSlicer #12495: Coloring with negative volumes not possible

CommandoBot, 2024-03-22:
### Description of the bug

I use cylinders and wooden plugs to put togehter my larger models that won't fit on a print plate. Therefore I place negative volumes which allign perfectly with the separated parts.
With my XL now I have the issue that I have a lot of tool swaps that are completely unneccesary and just result in higher print time and wear.

The surface of the negative volume always is the one of the main body. It is not possible to paint those surfaces nor does the slicer use the surrounding color.

[InternalColor.zip](https://github.com/prusa3d/PrusaSlicer/files/14720119/InternalColor.zip)

### Project file & How to reproduce

1. Place a regular geometry on the printplate
2. Add negative volumes to the object in a way that it forms open holes on the surface (like drill holes)
[image]

3. Paint the main object (coloring) in a way that one of the "holes" completely is surrounded by a single color
[image]

4. slice the model
[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.3 alpha1

### Operating system

Windows 10

### Printer model

Prusa XL 5T

CommandoBot, 2024-03-22:
The funny part in this example is, that the bottom of the hole actually is using the surrounding color, but then it changes to the main object color again ...

Jan-Soustruznik, 2024-03-27:
Hello, @CommandoBot , Thank you for your report. This is limitation of the current implementation of the multimaterial painting feature. If you want to have the surface of the negative volume painted same color is neccesary to select extruder with requested color to the model.

closed 2026-09-29 by prusaslicer-bot[bot]