prusa3d/PrusaSlicer #12497: Organic supports do not support themselves

siarczan6miedzi2, 2024-03-22:
### Description of the bug

In some cases layers of organic supports start mid air or at least are not supported enough. In the provided file the supports under the balls start with a very small layer which is not joined to other parts until about 4 mm of a print. This would definitely collapse mid-print. In some of my other trials with other number of models and some additional tweaks for artificial supports, these tiny layers of supports started as high as 3 layers above the raft, so completely mid-air.
This issue could be easily prevented by adding possibility of infill within organic supports, as for example Cura enables (to the best of my knowledge, this feature is sadly not present in Prusa Slicer)

### Project file & How to reproduce

[issue.zip](https://github.com/prusa3d/PrusaSlicer/files/14725805/issue.zip)

The zip file contains project file (.3mf) and exported G-code (.gcode)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Debian 12

### Printer model

Ender 3

neophyl, 2024-03-22:
Duplicate of many issues already raised.  Such as
https://github.com/prusa3d/PrusaSlicer/issues/12418
https://github.com/prusa3d/PrusaSlicer/issues/9984
https://github.com/prusa3d/PrusaSlicer/issues/9601
etc

Please search for existing issues before opening new ones.

Jan-Soustruznik, 2024-03-25:
Hello, @siarczan6miedzi2 , thank you for you report, but how @neophyl mension, issue already reported. I close issue as duplicate. Thank you, have a nice day

closed 2024-03-25 by Jan-Soustruznik