prusa3d/PrusaSlicer #12498: G92 E0 missing (gcode empty line)

vetoriando, 2024-03-22:
### Description of the bug

Print starts just fine, but at some point the line where should be the G92 E0 becomes empty. The printer starts the next layer retrating all the filament used on the previous layer.

Until layers are solid, everything is fine...
[image]
But as I'm using only one perimeter (0% infill)...
[image]

Other parts (variants of the same project) go a long way but start missing the G92 only at the end, taking 6h of print to reach the failing point. It becomes easier to see in the cuted part.

In this point I noticed a pattern. The problem begins when opposite contours touch each other.
[image]
[image]

### Project file & How to reproduce

Slice it with 0% infill and look for empty lines on the gcode. Should be G92 E0 there.
[Pistas.zip](https://github.com/prusa3d/PrusaSlicer/files/14726556/Pistas.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2+win64

### Operating system

Windows 10

### Printer model

Sethi S3

kvikindi, 2024-03-22:
I found I had to disable ramping lift in order to get this functionality working again. Your results may vary, but I think there might even be a fix in the latest alpha version? (2.7.3-alpha1)

vetoriando, 2024-03-22:
I cannot imagine the relation between ramping lift and Extruder feed, but it changed the gcode a lot and reading the code now looks like that it solves the problem.

Later I'll upgrade to version 2.7.3. I didn't updated yet because no fixes related to that was mentioned.

Jan-Soustruznik, 2024-03-25:
Hello, @kvikindi, Thank you for your report. Yes problem is caused by the Ramping Z lift feature. This issue is already fixed in 2.7.3-alpha1. I closed this issue as duplicate. https://github.com/prusa3d/PrusaSlicer/issues/12428

closed 2024-03-25 by Jan-Soustruznik