prusa3d/PrusaSlicer #12500: Ramping Lift does not occur between organic support structures

root-objectId, 2024-03-22:
### Description of the bug

v2.7.2 
Ramping Lift enabled
1.5mm max height
8* slope
I am having issues where it does not ramp lift between different organic support structures. It'll lift between the model and the first support, but then doesn't lift between the different support structures. On it's way back to the model after the supports, the toolhead collides with the structures, eventually failing the print.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Windows 11

### Printer model

Ender 5 Pro Octopi/Klipper

root-objectId, 2024-03-22:
Enforce lift over X travel distance?

SachCZ, 2024-03-25:
Hello @root-objectId, can you please provide a 3mf file? Thank you for your cooperation.

root-objectId, 2024-03-27:
GitHub doesn't seem to allow the posting of a .3mf file? How can I provide?

root-objectId, 2024-03-28:
Apologies, updated OP with file

SachCZ, 2024-03-28:
Thank you @root-objectId, sorry for not replying in time, but you seem to have figured it out yourself :rocket:. We will look into it.

SachCZ, 2024-03-28:
Hello @root-objectId, you have `Use firmware retraction` **checked** in the `Printer settings -> General`. This effectively disables any lifts. I agree that it is confusing, but it is intended. To enable the ramping lifts uncheck the firmware retraction and **make sure to use non 0 value** for the retraction setting in `Printer settings -> Extruder 1 -> Retraction -> Length`. If you have any further issues, feel free to let us know. I am closing this as it is not a bug nor a feature request. Thank you for your time and I wish you happy printing!

closed 2024-03-28 by SachCZ