prusa3d/PrusaSlicer #12502: Bambu 3MF

xayoz, 2024-03-23:
**Is your feature request related to a problem? Please describe.**
2.7.3 now opens a Bambu 3MF, which is fantastic, but only opens it the same as an STL.  Would love if it kept colour painting from models so that I could assign the changes to my MMU that were previously assigned to the Bambu owner who made the 3MF.

**Describe the solution you'd like**
Keep colour painting from Bambu and allow to select which MMU slot or XL toolhead to assign

**Describe how it would work**
Keep colour painting from Bambu and allow to select which MMU slot or XL toolhead to assign

**Describe alternatives you've considered**
AN/A

**Additional context**
N/A

ThorsReign, 2024-03-27:
This would price the superiority of the Prusa XL over Bambu printers for waste and speed of the same files could be opened and run with the same painting but without all of the purge and time swapping.

kelcyknapp, 2024-03-27:
I have a library of some 100-200 Bambu MMU 3mfs paintjobs that took me roughly 3 weeks to get painted that I will be unable to use with my PrusaXL arriving in a week, which is really depressing and will limit my use of that spectacular printer. This should be a thing already. We beg, Prusa.

lukasmatena, 2024-04-04:
It might be interesting to first describe how the problem came to existence.

The reason why color painting is lost when loading 3MF files from BambuStudio is because BambuLab forked PrusaSlicer and for some reason changed the key under which the painting data are stored. PrusaSlicer saves them at each triangle using `slic3rpe:mmu_segmentation` attribute (as it always had), while BS changed the key to `paint_color`. The format of the data and its location is AFAIK the same. I have no idea why the change was necessary, it seems to be a completely unneeded compatibility-breaking change. My impression is that the incompatibility you "beg" us to fix is there simply because BambuLab decided it should be there and made it so.

It is also worth mentioning that adding an XML tag without a namespace prefix (the `slic3rpe:`) is not allowed by 3MF Core Specification (see [here](https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#2331-xml-namespaces)). I will not claim that PrusaSlicer is perfect when it comes to following 3MF specs, but this change actually made it worse. (Despite the claims about adhering to 3MF Core Spec in the BS 1.8.4 release notes.)

Technically, we can still load the data by looking for either `paint_color` or `slic3rpe:mmu_segmentation` and treat both attributes equally. We will probably do that in the next release. However, I don't exactly like the idea of implementing support for things that our forks deliberately break...

lukasmatena, 2024-04-05:
The painted colors from BambuStudio should load in [2.7.4](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.4). Closing.

closed 2024-04-05 by lukasmatena

raistlinJ, 2025-08-09:
I created a converter here: https://github.com/raistlinJ/3mf_bambu2prusa
Tested with several models and it seems to work; even transfers painted objects.
Hope it helps.