prusa3d/PrusaSlicer #12505: Suboptimal tool path for bridging

kdobrev, 2024-03-23:
### Description of the bug

The first top layer start with weird tool path that causes lines printed over thin air without any anchor points.
[image]

The model is https://www.printables.com/model/791933-the-smallest-whistle

### Project file & How to reproduce

Project file is:
[Smol_Whistle.zip](https://github.com/prusa3d/PrusaSlicer/files/14733715/Smol_Whistle.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Windows 10

### Printer model

Sovol SV06 Plus

u89djt, 2024-03-24:
Confirming that the slicer starts mid-air, something that definitely shouldn't happen, and noting that you have a non-standard bridging setting that isn't necessary for this model so there's a simple workaround in this case.
(not affiliated with prusa)
Starts in mid-air
[image]
If you de-select "Thick bridges" under "Layers and perimeters" you get feasible extrusion.
[image]
[image]

Jan-Soustruznik, 2024-03-26:
Closed as duplicate https://github.com/prusa3d/PrusaSlicer/issues/11430

closed 2024-03-26 by Jan-Soustruznik