prusa3d/PrusaSlicer #12508: Missing G1 E-4 after Filament-/Tool-change M600 using Multimaterial with Singelextruder

Ventilo2, 2024-03-25:
### Description of the bug

After Toolchange/Filamentchange M600 there is a G1 E4 which produce some not wanted Filemant extrude.
What i want to do is to use multicolor on 1st and 2nd layer.

Printersettings - common 
- Extruder: 2
- Singelextruder with Multicolor: true

using Multimaterial painting for e.g. a Text on the first two layers 
(additional Feature Request: number of layers for multicolor adjustable. I put a gab on layer 3 to prevent too much Filamant changes)

i modify the gcode-file manually and it will work.

by the way the G1 E4 F2100 is to much, maybe a G1 E.8 F2100 like before is good too.

line 3980 and following:
M142 S36 ; set heatbreak target temp
M486 S0
G1 E-.8
G1 X117.641 Y108.756 F18000
G1 Z.2 F720
G1 E.8 F2100
...
Line 5417 and following:
M142 S36 ; set heatbreak target temp
M486 S0 
 ; <------------------   G1 E-4 F2100 is missing here !
G1 X105.7 Y110.3 F18000
G1 Z.4 F720
G1 E4 F2100 ; <------------------ this line will produce unwanted Filament, because there is a G1 E-4 missing before
...

----

Maybe use G-4 only for Flex TPU, like in the default Start G-Code
{if filament_type[initial_tool]=="FLEX"}
G1 E-4 F2400 ; retraction
{else}
G1 E-2 F2400 ; retraction
{endif}

### Project file & How to reproduce

[Bug files 3mf and gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/14737785/Bug.files.3mf.and.gcode.zip)

here are the lines i modify everytime i will use 2 or more color on first two layers 
as example i have a two color print.

line 2971: ; M600 ; <------------------------------------ removed, first Filament-/Tool-change not needed - mod by Mark

line 3970: G1 X117.641 Y108.756 F18000; <---------- add a jump to the next position before Filamant change - mod by Mark
line 3971: M600

**here begins the bug...**

line 5406 ; Filament-specific end gcode
G1 X105.7 Y110.3 F18000; <------------- add a jump to the next position before Filamant change - mod by Mark
M600
T0
M104 S250 ; set temperature
M900 K0.07 ; Filament gcode

M572 S0.055 ; Filament gcode

M142 S36 ; set heatbreak target temp
M486 S0
G1 E-4 F2100 ; <-------------- i added this line to prevent the Error   - mod by Mark
G1 X105.7 Y110.3 F18000
G1 Z.4 F720
G1 E4 F2100 ; <------------------ this line will produce unwanted Filament, because there is a G1 E-4 missing before

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2 win-64

### Operating system

Windows 10

### Printer model

MK4 IS

rtyr, 2024-03-25:
"single extruder multi material" is not meant for "M600 hack", it says you are using MMU-like hardware on your printer

If you don't want that 4mm toolchange retraction, set it to 0mm in extruder settings (you have to set it on every extruder).

closed 2024-03-25 by rtyr

Ventilo2, 2024-03-25:
i use the default settings for MK4!
And negative extrude is still missing, if i use the MMU or manually change the Filament.

rtyr, 2024-03-25:
You are not using default settings. You added more extruders, M600 to toolchange, etc. That value you are refering to is never used in standard settings.

Ventilo2, 2024-03-25:
okay, thank you. 
I will use to set the length to 0 and will wait to get the MMU3 ;-)

But Why is one time G-.8 and G8 and the second change only with G4 an no G-4?
I don't want an answer ;-)
have a nice day and thank you very much for your work.

rtyr, 2024-03-25:
Toolchange retraction is different thing, than standard retraction. It is supposed to retract filament on disabled/parked tool, typically on printers with more nozzles/extruders.

You may use `{if previous_extruder!=-1}M600{endif}` in toolchange script, instead of` M600`. It will remove that unwanted M600 at the beginning. You should also disable "single nozzle multi-material" parameter.

We may remove that unused 4mm value via configuration update, to prevent issues like this wich custom profiles.