prusa3d/PrusaSlicer #12515: Arrange behavior abnormal

TonFTP, 2024-03-27:
### Description of the bug

When clicking arrange, it centers them at the back of the print bed.  After slicing, returning to the editor view and clicking arrange, it centers them in the center of the print bed as expected.  Clicking arrange a 2nd time, will again center them at the back of the print bed until another slice is done.  It may be related to the wipe tower?

When it is in an abnormal state, the "Alignment" option in the right-click menu of the arrange button goes missing and returns after slicing.

### Project file & How to reproduce

Open or import STL, click arrange, slice, click arrange, click arrange again.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2+win64

### Operating system

Windows 10

### Printer model

Prusa XL

TonFTP, 2024-03-27:
Initial open:
[image]
After slicing and returning to the editor view:
[image]
After clicking Arrange once:
[image]
After clicking Arrange again:
[image]
After slicing and returning to editor:
[image]

u89djt, 2024-03-30:
The 3mf file you uploaded contains different objects to the one in your illustrative screenshots:
[image]

TonFTP, 2024-04-17:
I must have grabbed the wrong file, however, it happens with any file, it is not specific to the one I pulled screenshots for.

neophyl, 2024-04-17:
I think this is a duplicate of several other existing issues, for example https://github.com/prusa3d/PrusaSlicer/issues/11414 or https://github.com/prusa3d/PrusaSlicer/issues/11785

closed 2026-09-29 by prusaslicer-bot[bot]