prusa3d/PrusaSlicer #12516: 2.7.3 rc-1 makes seams on the vase

IgorOhrimenko, 2024-03-27:
### Description of the bug

The program makes a seam in the vase.
I was printing a vase cut in version 2.7.3 and noticed that the printer prints slowly at the "seam", after printing the seam is visible and the seam is tactile.
The same model cut in 2.7.2 has no seam.
In the preview of the slicer, the models look the same.
[image]
[image]
I shot a video with real printed models:
[2.7.2-vs-2.7.3.webm](https://github.com/prusa3d/PrusaSlicer/assets/24807060/790b0047-250c-4c66-9df4-5b5a6fc0dee2)

### Project file & How to reproduce

[2.7.2-cylinder.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14781082/2.7.2-cylinder.3mf.zip)
[2.7.3-cylinder.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14781083/2.7.3-cylinder.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2 / 2.7.3 alpha / 2.7.3 beta / 2.7.3 rc-1

### Operating system

Window 10

### Printer model

Qidi X-Plus II

IgorOhrimenko, 2024-04-12:
The same problem in the new 2.7.4.

IgorOhrimenko, 2024-04-28:
@vovodroid can your check this problem, please?

vovodroid, 2024-04-28:
Sorry, I have no cue in this algorithm.

closed 2026-09-29 by prusaslicer-bot[bot]