prusa3d/PrusaSlicer #12518: Cannot open Bambu 3MF with only primitive shapes

CableCatDK, 2024-03-28:
### Description of the bug

When opening a Bambu 3MF with only primitive shapes, I get this error:
- Error (Invalid 3MF format) while parsing '3D/3dmodel.model' at line 22

### Project file & How to reproduce

How to reproduce:
- Open FRYELtoFUGA_Bambu.3mf from [this zip-file](https://github.com/prusa3d/PrusaSlicer/files/14786842/FRYELtoFUGA_Prusa.zip).

Alternative way to reproduce:
1. In PrusaSlicer start a new project
2. Add a primitive shape
3. Save the project
4. Close PrusaSlicer 
5. Open the project BambuSlicer
6. Save the project
7. Try to open the project in PrusaSlicer agian.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.3-rc1

### Operating system

Windows 11

### Printer model

Prusa MK4

Jan-Soustruznik, 2024-03-28:
Hello, @CableCatDK , thank you for your report. We will try to fix in some next releases. Issue created: SPE-2207

Jony01, 2024-04-02:
I try to check `FRYELtoFUGA_Bambu.3mf` file by official [3mf verifier](https://github.com/3MFConsortium/3mf-samples/tree/master/verifier):
```3mfverify.exe FRYELtoFUGA_Bambu.3mf core_1.3.0.xsd production_1.2.xsd```
result:
```
3MF Verify
v1.0
(c) 2017 3MF Consortium

=======================================
3MFVerify started on file: FRYELtoFUGA_Bambu.3mf
=======================================
FRYELtoFUGA_Bambu.3mf opened successfully
53740 bytes read
Does FRYELtoFUGA_Bambu.3mf contain an OPC package?
 ----------------------------------------
Exception: ZIP archive contains unsupported data structures.
 ----------------------------------------
Usage: 3MFVerifier.exe <file.3mf> [<schema1.xsd> | <schema2.xsd>
Exception: 0
Verifying mesh.
ERROR: One or more errors occurred.
XML verification incomplete; undefined error found
----------------------------------------
3MFVerify complete: Failure
```

By the `.model` file in 3mf is required production extension:
xmlns:p="http://schemas.microsoft.com/3dmanufacturing/production/2015/06" --> [production_1.2.xsd](https://github.com/3MFConsortium/spec_production/blob/master/3MF%20Production%20Extension.md#appendix-b-3mf-production-extension-schema)
xmlns="http://schemas.microsoft.com/3dmanufacturing/core/2015/02" --> [core_1.3.xsd](https://github.com/3MFConsortium/spec_core/blob/master/3MF%20Core%20Specification.md#appendix-b1-3mf-xsd-schema)

closed 2026-09-29 by prusaslicer-bot[bot]