prusa3d/PrusaSlicer #12519: Wrong "Fixed a bug resulting in inadequate anchoring of bridging extrusions" in 2.7.3

Den2016, 2024-03-28:
### Description of the bug

Release 2.7.3 presents that inadequate bridge fixes have been fixed. This is not entirely true. 
First of all, the outer perimeters are extruded first, which may already be a geometry violation
[image]
[image]

Secondly - the actual print, although it looks better, is still far from perfect
[image]

[image]

### Project file & How to reproduce

[płytka_naprawcza_drzwi.zip](https://github.com/prusa3d/PrusaSlicer/files/14794510/plytka_naprawcza_drzwi.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.3

### Operating system

Windows 7 x64

### Printer model

Kingroon KP3S

neophyl, 2024-03-28:
Try turning thick bridges to OFF.  You get a better slice result by not using the legacy setting
With it =ON
[image]

With it =OFF
[image]

You can also experiment with changing the perimeters/infill overlap % and the bridge flow ratio slightly.

SachCZ, 2024-03-28:
Hello @Den2016, **thank you** for your feedback. **The mentioned bugfix does not claim to address this concrete case.** Bridges not sticking to bridge perimeters is a **complex problem** that depends not only on the PrusaSlic3r path generation and concrete slicing settings but also on the concrete printer used (its fan may cool a bit more/less etc.).

In this particular case it seems that just rotating the object would address a lot of problems, I thus suppose you are posting this as an example of an object where this particular difficulty arises. Besides the solutions that @neophyl so kindly provided you may also try to use the automatic support painting feature which in this case supports the bridge anchors better.

Your problem is **not** caused by a **bug in PrusaSlic3r** - I am thus closing this issue. **If you still think there is a bug let us know!** If you want to propose a solution, feel free to open a feature request, just make sure to **provide detailed description** of the issue and the proposed solution.

closed 2024-03-28 by SachCZ