prusa3d/PrusaSlicer #12520: Неадекватное управление вентилятором при печати нависающих периметров

Den2016, 2024-03-28:
### Description of the bug

В настройках задана скорость вентилятора в 100% при печати нависающих периметров и мостов. При печати мостов все нормально, скорость выдерживается стабильно. Но вот при печати нависающих периметров наблюдаю постоянное изменений скорости вентилятора на каждом отрезке периметра, причем скорость скачет от минимальной из настроек динамической скорости охлаждения до заданной для печати мостов. Более того - на каждый отрезок периметра выставляется от 2 до 5 команд установки скорости ( отчего зависит - я еще не понял). При этом вторая команда скидывает до 35%, затем идут две, а то и три команды на 100%. В результате, на каждом сегменте вентилятор притормаживает и снижает эффективность охлаждения. Дублирование этого набора команд увеличивает объем g-code. Например, вот этот сегмент дуги имеет 56 частей, каждая из которых сопровождается установкой скоростей вентиялтора в количестве четырех команд.
[image]

### Project file & How to reproduce

The files can be taken from #12519

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7 all

### Operating system

all versions

### Printer model

all printers

neophyl, 2024-03-28:
That 3mf file has dynamic fan speed enabled.  
[image]

With one end set to 100% and the other set to 0% its looks like its doing what the tooltip says and working it out dynamically between 0-100%.  So that's why you have so many M106 lines in there.  Exactly as configured.  Turning that off reduces the number of M106 commands by a fair amount but it still jumps between 89.25 and 255.  There are just less commands overall.  
Seeing as 89.25 is 35% it looks like its constantly jumping between your minimum fan speed and the bridge fan speed for those perimeters (layer 41).  2 consecutive fan speeds like that looks like a bug to me.
The slice preview is only showing the 100% as the pairing has the 255 value last.  

For the actual bridges that seems to stick at the 100%/255 value.

u89djt, 2024-03-29:
Is there anything relevant here? https://github.com/prusa3d/PrusaSlicer/issues/11981

closed 2026-09-29 by prusaslicer-bot[bot]