prusa3d/PrusaSlicer #12521: "Allow downloads from Printables" no longer working

Stooovie, 2024-03-28:
### Description of the bug

"Allow downloads from Printables" is enabled but I still get "PrusaSlicer received a dowload request from Printables but it's not allowed." Disabling, restarting, reenabling didn't fix.

### Project file & How to reproduce

1) click the PrusaSlicer button on a Printables model

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.3

### Operating system

MacOS Sonoma 14.4.1

### Printer model

Ender 3

mix579, 2024-03-28:
I can confirm this is not working anymore on my M1 MacBook Air MacOS 14.3.1

lukasmatena, 2024-03-29:
@Stooovie
> never had it before, not even the 2.7.3 betas

Can you please confirm that the issue is not present in [2.7.3-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-beta1)? Thanks.

FidelCapo, 2024-03-29:
I can confirm this issue. The bug happened between 2.7.3.-alpha and 2.7.3-beta versions. We will try to fix the bug.

Stooovie, 2024-03-29:
> @Stooovie
> 
> > never had it before, not even the 2.7.3 betas
> 
> Can you please confirm that the issue is not present in [2.7.3-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.3-beta1)? Thanks.

I was wrong, it does happen as well in beta 1.

salmander, 2024-03-30:
I experienced this as well. Try unchecking that checkbox and it worked for me.

Stooovie, 2024-03-30:
> I experienced this as well. Try unchecking that checkbox. That did the trick for me.

LOL yes you're right. Unchecking "Allowed" in fact allows direct downloads :)

ulab, 2024-04-02:
The issue still is in the 2.7.3 (2.7.3+MacOS-arm64) release.

K1ngjulien, 2024-04-05:
Same here on linux, for me the checkbox even disappeared lol. it worked a week ago so i guess an update broke it?
[image]

PrusaSlicer 2.7.3

ulab, 2024-04-05:
This is fixed in 2.7.4.

lukasmatena, 2024-04-05:
Thanks for your reports. Fixed in [2.7.4](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.7.4), as @ulab mentions. Closing.

closed 2024-04-05 by lukasmatena

Ethan-Ming, 2024-11-17:
i know this issue is closed but I got it again in 2.8.1, simply time out, 
also in mac os

[image]