prusa3d/PrusaSlicer #12524: Crashing during slicing.

JosephSchafer, 2024-03-29:
### Description of the bug

PrusaSlicer crashes on 2.7.3 when slicing.

PrusaSlicer
Version:   2.7.3+win64
Build:     PrusaSlicer-2.7.3+win64-202403280959

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 11 (build 22631), 64-bit edition
Total RAM size [MB]: 34,160MB
OpenGL installation
GL version:   4.6.0 (4.6.0 NVIDIA 551.76)
Profile:      Core
Vendor:       NVIDIA Corporation
Renderer:     NVIDIA GeForce RTX 4090/PCIe/SSE2
GLSL version: 4.60.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
[... 367 lines cut ...]
Report Id: c96926b8-f157-4730-8eff-3c0bce93e4f7
Faulting package full name: 
Faulting package-relative application ID: 

### Project file & How to reproduce

Seems like any attempt to slice is crashing PrusaSlicer.
[Crossbar Wall Mount.zip](https://github.com/prusa3d/PrusaSlicer/files/14806085/Crossbar.Wall.Mount.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

22631.3296

### Operating system

Windows 11 Pro 23H2 OS Build: 22631.3296

### Printer model

Mk3+

Jan-Soustruznik, 2024-04-02:
Hello, @JosephSchafer, We cannot evoke your reported crash. Could you check if the shared 3mf files is correct? Or Could you try to reinstall the slicer? Thank you, have a nice day

profOnno, 2024-04-03:
Also got a crashing slicer. v2.7.3 on linux. It will crash in `libslic3r/GCode.cpp:4045`, I commented the assert, now it works. The `if` function, after the assert, seems useless when the assert before make's the crash. Correct me if I'm wrong.

Don't know if this is the same problem as @JosephSchafer has.

Jan-Soustruznik, 2024-04-04:
> Also got a crashing slicer. v2.7.3 on linux. It will crash in `libslic3r/GCode.cpp:4045`, I commented the assert, now it works. The `if` function, after the assert, seems useless when the assert [...]

Hello, @profOnno , did you use the same 3MF what @JosephSchafer share?

profOnno, 2024-04-04:
I just tried  the 3MF.
And it works, with my modified code (the removed assert). The `gcode`  generation take some time and makes my computer work. The crash can also depend on the slicer settings, so not sure if we've got the same problem. I'm printing with a .6 mm nozzle (slicing won't match .4 mm nozzle).

[image]

The question about the `assert` and the `if` statement afterwards still stands.

I did not change the code back to see if it will cause the problem. Maybe @JosephSchafer can run the slicer from the command line and it should have the stack trace (maybe?).

profOnno, 2024-04-04:
Hmm trying now, with assert on. Last time my settings were maybe of.
But I got the crash.
```
prusa-slicer: /home/teknomen/3d/prusaSlic3r/PrusaSlicer/src/libslic3r/GCode.cpp:4046: std::string Slic3r::GCodeGenerator::_extrude(const Slic3r::ExtrusionAttributes&, const Path&, std::string_view, double): Assertion `p != prev' failed.
Aborted (core dumped)
```
@JosephSchafer, If we can reproduce the crash, we can improve the software and prevent the crash (give a decent warning/ error message). If it was up to me, I wouldn't close the issue (If it's reproduceable), but I'm not the maintainer of the project.

JosephSchafer, 2024-04-04:
Makes sense.  Just don't want volunteers volunteering on my account.  My
quest for a well-organized garage via 3D printing has been furthered. ;)

closed 2026-09-29 by prusaslicer-bot[bot]