prusa3d/PrusaSlicer #12526: Prusa slicer moves perimeter when I add mouse ears

uzytkownik, 2024-03-29:
### Description of the bug

When I try to add mouse ears (here is a artificial example as I cannot share original files) the perimeter of main object 'hugs' the ears.
[Shape-Box.zip](https://github.com/prusa3d/PrusaSlicer/files/14807794/Shape-Box.zip)

### Project file & How to reproduce

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

NixOS Unstable

### Printer model

Prusa Mk3S+

u89djt, 2024-03-29:
(not affiliated with Prusa)
That's an intended feature of the elephant's foot compensation function. The first layer is pulled inward by the distance specified in the advanced setting just at the exposed edges so they don't squish too far outwards.
Where you've added a body continuous with the original outer edge, it's no longer an exposed edge and so is not pulled back with the rest, and so it ends up hugging your ear.
[image]
Set the compensation to 0 and you'll see the perimeter doesn't deviate as it goes past the ears. With compensation set to 0.2, show layer 2 and you'll see how far the compensation shifts and where.

MrBlackmidi, 2024-03-31:
makes sense

choffee, 2026-01-09:
I think this is related to #2954

closed 2026-09-29 by prusaslicer-bot[bot]