prusa3d/PrusaSlicer #12527: Measuring Tool isnt working correct

aclucca, 2024-03-29:
### Description of the bug

A measuremnt between a plane and a circle should be done in nomal direction!

### Project file & How to reproduce

[Desktop.zip](https://github.com/prusa3d/PrusaSlicer/files/14807848/Desktop.zip)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Windows 10 64bit

### Printer model

PRUSA MK4

u89djt, 2024-03-29:
Workaround for this specific example: if you select the circle, then the face, then the centre of the circle, it gives you the normal distance.
[image]
(not affiliated with Prusa)

aclucca, 2024-03-29:
> Workaround for this specific example: if you select the circle, then the face, then the centre of the circle, it gives you the normal distance. [...]

perfect :-) thank you for the workaround :-)

u89djt, 2024-03-29:
Turns out you can also do it in the order face, circle, point, /but/ circle, face, point lets you then continue to select another face and immediately measure the normal distance to that one.

u89djt, 2024-03-30:
@aclucca I've explored more and concluded (opinionated user reporting in) that the in-slicer measurement system is impressively solid. You might be interested to see this measurement between a hole and a plane at a free angle. This is the choice of dimension that would be shown between a plane and a point in other design packages. 
[image]
Select the edge of that plane, and you again get a normal measurement
[image]
[measurement toy.zip](https://github.com/prusa3d/PrusaSlicer/files/14811648/measurement.toy.zip)

closed 2026-09-29 by prusaslicer-bot[bot]