prusa3d/PrusaSlicer #12528: slicing time jumped 20x times

FTIJPI, 2024-03-30:
### Description of the bug

The slicing result is 1 hr for to print a large flat letter with 1 mm height.
But the slicing result is 20 hrs for to print a large flat letter with 1.5 mm height.

As shown int the 2 pictures. I reloaded this file, open and closed the slicer and restarted my PC. The result is the same.

[image]
[image]

### Project file & How to reproduce

I'm just reporting this as a user

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.0 based on slic3r

### Operating system

Win 10

### Printer model

Neptune 3 Max

u89djt, 2024-03-30:
"How to reproduce" is an unmissable step in reporting issues. For your problem, since you don't know why it is happening, and can't know what is important, save each example to a 3mf file, compress those in a zip file and copy that into your post. Without doing that, you're asking people to guess or magically know what's happening, essentially.
/By sheer luck/ this time, there's enough information to make an educated guess.
You've cropped the screenshots, but it looks like you have activated ironing in the long duration print. That does take a very long time compared to normal layer deposition. You also seem to have disabled top and bottom solid layers and chosen a single perimeter in the quicker print. That will save time. So you're comparing one of the longest duration print operations (ironing) with one of the quickest ways to print (no bottom or top solid layers and a single perimeter).
So, you've explicitly told the slicer to use a very different combination of settings that will necessarily take all that extra time.
(not affiliated with Prusa)

J3r0github, 2024-03-31:
I'd need a project file to attempt to attempt reproduce this myself. You can export the project files for both from _File -> Save project as_

@u89djt however made pretty good points, in that in your cropped screenshots it does indeed look like you're comparing apples to oranges.

closed 2026-09-29 by prusaslicer-bot[bot]