prusa3d/PrusaSlicer #12529: M73 is not emitted for Stealth Mode on the MK4

fmorgner, 2024-03-30:
### Description of the bug

The MK4 firmware 6.0.0 RC-1 supports Stealth Mode. However, PrusaSlicer 2.7.3 does not seem to emit the appropriate `M73` lines for the Stealth Mode time estimation (see https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3871). This might be an oversight, or simply something that has not been released, since the MK4 firmware 6.0.0 is not in stable release yet.

### Project file & How to reproduce

1. Create a cube in PrusaSlicer
2. Slice for MK3
3. Export gcode
4. Slice for MK4
5. Export gcode

The MK3 gcode contains `M73` commands for the Stealth Mode, the MK4 gcode doesn't.

Attached [is a ZIP](https://github.com/prusa3d/PrusaSlicer/files/14813338/cube-gcode.zip) containing the gcodes generated using the procedure outlined above.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.3.7-linux

### Operating system

Linux

### Printer model

MK4

rtyr, 2024-03-31:
We are aware of that. Stealth mode will be enabled later via configuration update, likely when final FW 6.0.0 is released.

fmorgner, 2024-03-31:
Thanks! I'll close this issue in that case.

closed 2024-03-31 by fmorgner