prusa3d/PrusaSlicer #12530: Excessive streaks/lines in vase mode

Velaxtor, 2024-03-31:
### Description of the bug

Excessive lines now appear near protruding features in vase mode. Layer height shown here is .20mm
[image]
The issue can be somewhat alleviated by choosing a (much) higher detail setting at 0.05mm:
[image]
However, when doing vase mode in 2.7.2 this problem does not appear, or significantly better 0.20mm:
[image]
I find it very unlikely that this problem isn't apparent with many other (vase mode) models as well. I tried making various changes to the 3d model here, and nothing solved it or made it better in any noticeable way.

### Project file & How to reproduce

[spaceinvaders3.zip](https://github.com/prusa3d/PrusaSlicer/files/14813973/spaceinvaders3.zip)
I've included the project file for the 2.7.3 version since that's where the issue is. The way to reproduce this bug, or problem, is simply by having a model with protruding features in vase mode. It shouldn't matter much how sharp the angle is on those features. At least i seemed to get the same results no matter what way i did it.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.3

### Operating system

Windows 11

### Printer model

Prusa Mk3s+

u89djt, 2024-03-31:
It looks like the slicer follows a standard non-vase regime to establish extrusion segments, then modifies those existing segments to follow a very gently rising spiral. When the vertically neighbouring segments differ starkly, it's got to take some kind of average path between the available endpoints.
If this is what's happening, I guess breaking every horizontal segment at any breaks immediately above and below it before performing the spiral transformation would lead to a result that would be different and possibly better. This would involve writing an additional planar slicing result during that breaking process. This would still inevitably smooth features vertically somewhat. Maybe it could be an option to tick in the settings.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

For the crowd gathered, it's nothing to do with the model being supplied as an open shell with an overlap all the way up one corner.
(not affiliated with Prusa)

u89djt, 2024-03-31:
Smaller illustrative model
[image]
[vase mode extrusion interpolation.zip](https://github.com/prusa3d/PrusaSlicer/files/14816429/vase.mode.extrusion.interpolation.zip)

u89djt, 2024-03-31:
Based on pervceiving a need to break down the extrusion segments further, I tried adding fuzzy skin with a tiny thickness and short point distance and it does what we're hoping for pretty well. There's a spot where it's got a mean streak whatever values I try, and that appears in the planar slice before conversion to a spiral.
Settings:
[image]
Planar slice:
[image]
Converted to a spiral:
[image]
3mf file:
[fuzzy spaceinvaders3.zip](https://github.com/prusa3d/PrusaSlicer/files/14816968/fuzzy.spaceinvaders3.zip)

Velaxtor, 2024-03-31:
@u89djt thank you so much for going so methodically and thoroughly through the issue, i really appreciate it. Also nice work on identifying the underlying issue. I figured it was something like that, causing it, since the new build has a different way of dealing with smoothing in vase mode. 

I'll see if i can get to try printing the model in the coming days with the fuzzy fix/workaround. It looks just like the result from v2.7.2 so should be good! I do hope there will be a tick box for the smoothing in the future still.

u89djt, 2024-04-01:
You're very welcome! I flew at it for the fun of it. Let's wait for a dev's opinion.

closed 2026-09-29 by prusaslicer-bot[bot]