prusa3d/PrusaSlicer #12533: New setting: first layer extrusion multiplier

cwm9cwm9, 2024-03-31:
When printing on a textured surface, the z-offset of the printer must be adjusted so that the layer thickness is actually thicker than set in the slicer or the lines printed come out too thick.  This is because the texture takes up space that would normally be filled by extruded filament, and the excess plastic squishes outward.  This is a minor headache for those of us trying to print precisely dimensioned parts --- the first layer is often .25mm to .3mm thick instead of .2mm thick and/or there is substantial elephant foot.

A super easy solution to combat this would be a second extrusion rate multiplier that is only used on the first layer.  (The filament multiplier would still be used --- this would simply be an extra factor.  If your filament extrusion rate is .98 and your first layer rate is .75, then your overall rate is .98*.75 = .735.

The elephant footing can partially compensate for this, but not totally.  Because there is so much extra plastic that is extruded relative to what is needed, the amount of elephant foot partially depends on the feature size being printed.  A single line has much less than does a large base.

You can always adjust the z-offset of the printer to make extra space, but again, the print then comes out .05 to .1 mm thicker than intended.  (This may not seem like much, but if you are trying to print, say, a 1mm thick gasket, then 1.1mm is 10% too thick!)

In advanced, under the elephant foot setting, simply include:
"First layer extrusion multiplier:" with the default set to 1 and a note that reads, "the extrusion rate of the first layer will be reduced by this factor.  This can be used to compensate for textured print beds."

Thanks!

u89djt, 2024-04-01:
I think you might be looking for first layer extrusion width?
[image]
(not affiliated with Prusa)

cwm9cwm9, 2024-04-01:
No, I don't think so; that would also alter the tool path so the lines would be closer together.

rtyr, 2024-04-02:
I am closing this as a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/9719 and https://github.com/prusa3d/PrusaSlicer/issues/10470.

closed 2024-04-02 by rtyr