prusa3d/PrusaSlicer #12539: Z lift (retract_lift) is missing on layers above the first

vnd, 2024-04-02:
### Description of the bug

Z lifts are missing on all of the layers except for the first one. Models are pretty much default PrusaSlicer settings with very few parameters changed (haven't tried to print them, just analyzing g-code at this point).

In case of a trivial model without infill (testline), just perimeters, somehow enabling thin wall detection brings lifts back:
[image]

For another model with infill (sphere, created largely by just "Add Shape" -> "Sphere") , thin wall detection has no affect: However for this model, z lifts are there in 2.4.2.
[image]

### Project file & How to reproduce

[missing-lift.zip](https://github.com/prusa3d/PrusaSlicer/files/14842980/missing-lift.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.3

### Operating system

Windows 11

### Printer model

default (Original Prusa MINI)

neophyl, 2024-04-02:
Your profile has filament over rides enabled.  The filament is set to not retract if travel is less than 3mm.
[image]

With the size of the object you are printing which is small you aren't getting many of them due to the short travels. There are still a few though.
 If you scale up your object to make more of the travels longer than 3mm and look at the preview you can see it is lifting.
[image]

vnd, 2024-04-03:
Thanks, you're the best :) Spent some hours debugging this to no avail (was a PrusaSlicer fork but I thought I caught the problem with vanilla slicer).

I guess there might have been a bug in 2.4 where retract_before_travel was not (always) taken into account, but it's probably not important at this point.

closed 2024-04-03 by vnd