prusa3d/PrusaSlicer #12543: Default profile export names editor feature

ChrispyBaconIsGood, 2024-04-03:
I would like the software to be able to include information such as the filament weight used in the print I am selecting, and exclude useless information for me (printer model, I only own one!). It is frustrating having to sift through custom profiles, or edit each of my profiles individually, and I worry that the custom profiles are not up to date.

Interface could be in settings, a simple check box system, or list of items (include/exclude, order), to edit how the default profiles name the g.code exports.

I can then uncheck the "printer model" box, and check the "filament used (weight)" box, a much more streamlined process and much more friendly to the basic user.

I understand the concern around being able to edit default profiles, but this is incredibly low-risk to the user. If they mess it up, all they get is a weirdly named file which one quick google search will fix.

Tupson444, 2024-04-05:
Yes, the "output filename format" could be moved from Print Settings to general settings. I have many print profiles, but file name settings are the same in all. If I need to change something I would need to change it in all the profiles, so it would be more convenient if I could change it on one place.

However, to avoid braking current functionality, the Print Settings should have an option to override filename if needed (so current profiles would work the same, but if "filename override" is empty in the Print Settings, the global one would be used instead).

Tupson444, 2026-09-18:
still relevant