prusa3d/PrusaSlicer #12544: "Internal error: vector too long" when right click; select "fill bed with instances" of the provided object

u89djt, 2024-04-03:
### Description of the bug

Open as project "maximum pin print.3mf".
Right click on the merged group of objects on the plate and select "Fill bed with instances".
Popup error dialog appears as follows "PrusaSlicer Error Internal error: vector too long [OK]":
[image]
Click on OK and the program closes.
Objects taken from https://www.thingiverse.com/thing:243278/files by [Emmett Lalish](https://www.thingiverse.com/emmett)
Posted here in compliance with https://creativecommons.org/licenses/by-sa/3.0/ No changes were made to the object meshes; they are spatially distributed differently.

### Project file & How to reproduce

[u89djt240403.zip](https://github.com/prusa3d/PrusaSlicer/files/14852479/u89djt240403.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.3+win64-202403280959

### Operating system

Edition	Windows 10 Home Version	21H2 Installed on	‎16/‎06/‎2021 OS build	19044.3086 Experience	Windows Feature Experience Pack 1000.19041.1000.0

### Printer model

Original Prusa MK3S with no modifications or upgrades

u89djt, 2024-04-03:
I had a crazy offset from bed value. I stupidly just changed it wituout noting it. I'll search for the value to get the crash back and post it.

corywagner, 2024-05-21:
I just had the same error.  It happened during slicing.  Happens everytime on the same model.  Attached.  
[The castle J4.zip](https://github.com/prusa3d/PrusaSlicer/files/15395957/The.castle.J4.zip)

u89djt, 2024-05-22:
Hi @corywagner I've loaded the stl you provided into prusaslicer and sliced with Prusa XL defaults and it doesn't throw any errors. If you can post the 3mf file, I'll slice that and report back.
We should note that you've had the same error /message/, not necessarily the same error. It's a tedious fact in software engineering that most error messages are very, very specific about what's happening in the code at one instant and depressingly vague in relation to anything a user might care about :) It could be the same bit of code giving up, though. Including tracing information slows the code down, so in this instance, the engineers have chosen not to gather information about where in the code the problem happened. Giving them the 3mf will enable them to run the code in a debugging environment to find it. Or they might be familiar with the problem already and it's just very rare and a low priority :)

u89djt, 2024-05-22:
@corywagner I just ran it with the default MK3S profile and got the error. So it happens with MK3S but not with XL 5 tool input shaper 0.4 nozzles. Sorry, I should have thought more. But we shouldn't leave the devs wondering. 
Here's the castle in the MK3S profile: 
[MK3S The castle J4.zip](https://github.com/prusa3d/PrusaSlicer/files/15403410/MK3S.The.castle.J4.zip)
Here's the error dialog:
[image]
Hit OK on the error dialog and we see:
[image]
and if I hit the undo arrow, I see this:
[image]

u89djt, 2024-05-22:
@corywagner If you hit the arrange button before slicing, it succeeds in slicing it. You get a print stability warning because of all the overhangs, but it gives a printable result.

corywagner, 2024-05-22:
sorry, didn't work .  Same error.  I've included the 3MF file as requested
[The castle B4.zip](https://github.com/prusa3d/PrusaSlicer/files/15409754/The.castle.B4.zip)

corywagner, 2024-05-22:
BTW, u89djt, thank you very much for your help and insight.

u89djt, 2024-05-23:
Aha, details to chase! You've posted B4 there, and the one you had before was called J4. Are those from exactly the same stl file?

corywagner, 2024-05-23:
same model, different file name.

closed 2026-09-29 by prusaslicer-bot[bot]