prusa3d/PrusaSlicer #12546: Speed up GCODE generation if Start/End changes

lunatic1972, 2024-04-03:
**Is your feature request related to a problem? Please describe.**
Utilizing klipper I find myself once in awhile changing the gcode in the "Start G-Code" section. Once you do this, it requires it to re-generate all of the gcode again.

basically this is my klipper start macro:
PRINT_START INITIAL_BED_TEMP={first_layer_bed_temperature[0] - 5} BED_TEMP=[first_layer_bed_temperature] EXTRUDER_TEMP={first_layer_temperature[0]} LOADMESH=0  ENABLE_SKEW=0 FILAMENT_TYPE="[filament_type]" MOVENOZZLEX=15 PAUSE_TIME=3;

I change the PAUSE_TIME to a different value depending on PLA, PET OR ABS (Wait time for heating up)

**Describe the solution you'd like**
What if the start/end gcode was in an array for example so that when you re-sliced your file the only thing it would have to do is re-generate the array of "start g-code" rather than the whole thing.

The end could be done the same way.
Array[0]  "Start G-code"
array[...] "all the rest"
array[...] end gcode

I believe it would speed it up immensely.

gudvinr, 2024-04-06:
It won't work like that because start/end gcode can contain macro. 
New macro language can define variables that later used in oyher generated gcode.

lunatic1972, 2024-04-07:
I kind of thought about that and I've never used it but I guess there are people that could use it that way in the start/end. Thanks for the insight, I figured I would ask. You can close it then

gudvinr, 2024-04-09:
I am not PS developer so can't close issues. But you can close your own issues yourself.

closed 2026-09-29 by prusaslicer-bot[bot]