prusa3d/PrusaSlicer #12551: Top infill pattern monotonic has bug

atmacali, 2024-04-06:
### Description of the bug

When i use monotonic top infill it makes most of the part right to left but there is missed regions. There is lines on right side of the nozzle and it ruin the lines that produced earlier.

Cura has no issue like that pictures attached.

### Project file & How to reproduce

[prusa_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/14892148/prusa_bug.zip)
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.4+win64

### Operating system

win 10

### Printer model

ender 3 pro

u89djt, 2024-04-06:
Hi there, the horizontal slider at the bottom of the screen animates the progress of an individual layer. The vertical slider at the right of the screen selects a view of a whole layer. I think you moved the layer progress slider when you needed to move the layer selection slider.
This is a screenshot of the same situation you show in your first screenshot:
[image]
If you grab the slider at the bottom of the screen (shown here in a highlight) with your mouse, you can move it left and right, and the tool head indicator (the other highlight up and left over the object) will follow its extrusion paths around the layer backwards and forwards. You can watch the gaps that you're looking at being filled in as they will be in the physical print.
If you move the vertical slider instead, the tool indicator disappears and you just see whole layers being exposed, without gaps where material has yet to be laid down.
(not affiliated with Prusa)

u89djt, 2024-04-06:
"monotonic lines" looks closer to the cura result but it still does it as an old school flood fill. That's really interesting.
(random interested user)

closed 2026-09-29 by prusaslicer-bot[bot]