prusa3d/PrusaSlicer #12553: Fuzzy skin modifier gone in 2.7.4

nunofgs, 2024-04-06:
### Description of the bug

When I right-click an object I'm no longer able to add a modifier just for that part, specifically the fuzzy skin modifier. I'm now forced to enable fuzzy skin on the entire object by going to "print settings"

### Project file & How to reproduce

[scoop.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14894244/scoop.3mf.zip)

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

macOS 14.1.1

### Printer model

Prusa MK4

neophyl, 2024-04-06:
Still there for me with 2.7.4 on windows.  
[image]

[image]

You are in Expert mode aren't you ?  If you aren't then then bottom Add Settings option isn't visible.

Unless its another one of those MAC Os specific bugs like the ticked/unticked option for download from printables

nunofgs, 2024-04-09:
> You are in Expert mode aren't you ?

🤦 I wasn't. I never change it from expert mode. Dunno how I missed it. Thanks!

closed 2024-04-09 by nunofgs