prusa3d/PrusaSlicer #12562: Print settings, Filament and Printer scrolldown menu problem

Makak007, 2024-04-08:
### Description of the bug

This is a minor user interface issue.
You can not scroll up/down in Print settings, Filament and Printer scrolldown menus with mouse centre scroll wheel anymore - navigation in these menus is clumsy now. This occurred a few previous updates ago and still persists in 2.7.4 version.

### Project file & How to reproduce

[Trashbin Label_small_V05.zip](https://github.com/prusa3d/PrusaSlicer/files/14903904/Trashbin.Label_small_V05.zip)
[prusa-slicer_2024-04-08_11-54-17.zip](https://github.com/prusa3d/PrusaSlicer/files/14903906/prusa-slicer_2024-04-08_11-54-17.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Win7

### Printer model

MK3S+

neophyl, 2024-04-08:
Works for me.  I tried it on windows 10 and windows 7.  You do have to select the dropdown field but once its opened I can scroll up and down with the centre mouse wheel with no problems.  In all of the fields.

I also tried v2.5.0 as well as 2.7.4 (on my Win 10 machine) and the operation is the same in both versions.

Makak007, 2024-04-09:
Thank you for checking into this.
I tried to uninstall and install instead of upgrade but this did not solve the problem either.
I think I will have to live with this for now.

DoubleStrike, 2024-04-22:
This may need to be phrased differently.  The mousewheel DOES NOT work when the cursor is over the dropdown bar itself.  But if you move the cursor down onto the opened up part of the drop-down, then the mousewheel DOES work.  At least for me.

Maybe the cursor position is what is tripping you up.

Makak007, 2024-04-23:
Unfortunately this is not the problem, I tried to position my cursor carefully. Unfortunately there is no scrolling other than that dragging the slider up/down (orange in your printscreen, that is).

Apologies if I did not express myself clearly enough.
What I meant was when I click on the drop down menu, window (green area) with profiles opens with the scroll bar (orange) on the right edge of the window. Then I position my mouse cursor within the area of the window (green area) and scroll the mousewheel and nothing happens. I also tried to position the cursor over the scroll bar (orange) but that did not work either. I have only one option and that is to drag the scroll bar by mouse pointer. Even in that case the window demonstrates funny behavior and randomly(???) closes so I have to reopen it and select proper profile. Slow, annoying but doable after all.

My mouse scrolls perfectly in other applications, it is just within the PrusaSlicer where I observed this weird kind of malfunction.

Makak007, 2024-06-12:
Hello,
just checking if anyone heard of some progress in this issue.
I know I am not the only one, there is a similar (now closed) thread [Scrolling with mousewheel does not work](https://github.com/prusa3d/PrusaSlicer/issues/12564#top)
#12564 
it's a small but nagging thing :-)

Makak007, 2024-06-12:
I also to refer to neophyl's note about mouse HW/SW (in the other thread No. 12564).
I am using Logitech MX Vertical mouse, I also tried several USB wired mice - generally any mouse I could get my hands on but none of them can scroll in PrusaSlicer. 
No fancy or customized drivers though. It is really only in PrusaSlicer where scrolling does not work.

Kooshballer, 2024-12-24:
I am having the same issue with my prusa slicer and the mouse wheel not scrolling on any menu

closed 2026-09-29 by prusaslicer-bot[bot]