prusa3d/PrusaSlicer #12563: Bed size issue

dabear233, 2024-04-08:
### Description of the bug

Trying to print a project where the item size is exactly 220x220x70 on a Prusa MK4.  Receiving error that "an object is outside the print area".  This same item printed/sliced fine at 2.6. To get around this I needed to alter the bed size to 211 for the Y axis.  Printer has a print volume of 250 x 220 x 210, not 249 x 219 x 209.  

### Project file & How to reproduce

[Hex Spool Box.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14907945/Hex.Spool.Box.3mf.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

MacOS 14.3.1

### Printer model

Prusa MK4

u89djt, 2024-04-08:
Maybe it was infinitesimallly rotated, so the excess dimension was too small for the available significant figures in the display and for the arrange algorithm to get the right kind of grip? Here it is straightened up:  
[reoriented Hex Spool Box.zip](https://github.com/prusa3d/PrusaSlicer/files/14908376/reoriented.Hex.Spool.Box.zip)
[image]
It does seem like it would be good for the arrange algorithm to recover from that. I tumbled it around with "place on face" until it sat right.
(not affiliated with Prusa)

u89djt, 2024-04-08:
OK so that got past this specific problem, but if I hit arrange on that, even with both settings in arrange at zero, it says it's ignoring the object that can't fit on the bed.
[image]

u89djt, 2024-04-08:
If you open the original 3mf file as a project, it loads and shows blue; not fitting on the bed as in OP's report. However if you then delete the object, then add the 3mf file as an object, so leaving the project environment live, it loads and shows green.

closed 2026-09-29 by prusaslicer-bot[bot]