prusa3d/PrusaSlicer #12565: Arrange the objects by height

uzytkownik, 2024-04-09:
**Is your feature request related to a problem? Please describe.**
Currently it looks like prusa slicer either arrage items disregarding height or it seems to push high items to boundaries. This is undesired as it increases travel times.

In one of my prints today I managed to decrease print time from over 4 hours to 2:30 just by grouping high objects together avoiding unnecessary travel between them.

**Describe the solution you'd like**
Adjust prusa arrange algorithm for FDM printing so that the the objects using each layer are close together.

**Describe how it would work**
I am not familiar with arrange algorithm so I don't know what modifications would need to be implemented.

u89djt, 2024-04-09:
@kubispe1 this is a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/11961 but @uzytkownik has been specific about it being an FDM priority. Not news to you I'm sure, but I haven't left the FDMiverse yet so didn't recognize that when I read 11961 originally.
(random interested user)

kubispe1, 2024-04-09:
Hi, thanks for reminding me about this task and notifying me. I'm adding it to our list of potential nice to have - tasks. Have a nice day, guys!

closed 2026-09-29 by prusaslicer-bot[bot]