prusa3d/PrusaSlicer #12569: Addon to create precise brim ears

eefweenink, 2024-04-09:
**Is your feature request related to a problem? Please describe.**
Brim ears are used to prevent a print from warping. It is difficult to place the brim ears exactly. 
When they overlap the object to be printed, the effect is very visible in the print. 
When they are not close enough, it is complete useless to put a brim ear. 

**Describe the solution you'd like**
Playing around with the possibilities, I found a way to make them very precise: 
- Create a negative first layer copy of the object to be printed. 
- Scale it down so the size is 100%-minus a compensation for the elephant-compensation. 
- Save in galery.
- Put object on plate. 
- Place brim-ears (as instances)
- Add a negative part to the brim-ears, using the scaled negative first layer from the galery. 
- Adjust the negative part exactly to the placement of the object. 
DONE. The brim-ears will fit exactly. 
See https://www.printables.com/model/681785-precise-brim-ears-preventing-warp for an explanation with images. 

I think it should be quit easy to program this as an " Add brim-ears" in prusa-slicer. 

Happy printing, 

Regards, Eef

wafflecart, 2024-04-10:
“When they overlap the object to be printed, the effect is very visible in the print.
When they are not close enough, it is complete useless to put a brim ear.“

I think I have the solution for you it results in the discs actually forming part of the first layer but not so much they are hard to take off and when they are removed you cannot tell they were there.

- place your part on bed
- Right click add shape from gallery and add the “helper discs”
- copy and paste as many as needed and position
- Once finished and this is the key! Select all the instances (the part and all helper discs) in the right hand menu with the shift key then right click and merge.

[image]
[image]

Is that what you need? If not ignore me haha

wafflecart, 2024-04-10:
Also agree with your overall comment, a good feature in the slicer would be a button you click to add helper docs, mouse ears, brim ears whatever you want to call them. 

Could be done by the slicer detecting sharp corners and adding discs to the center parts of those corners.

eefweenink, 2024-04-10:
> * Once finished and this is the key! Select all the instances (the part and all helper discs) in the right hand menu with the shift key then right click and merge.
> * Once sliced you will see the bottom layer and all discs become one!
If I would have known that, I would probably not have suggested the feature. 

But there is a difference. Not much, but might give the "finishing touch" 
1. First layer, simply put on top of eachother
[image]

2. First layer, merged (your solution!)
[image]

3. First layer, my solution. (More work, that is why I suggested the feature)
[image]

Regards, Eef

Snuff1eupagus, 2024-04-27:
Perhaps this will help.

https://www.youtube.com/watch?v=dhsbMmosbLU&t=75s

eefweenink, 2024-04-27:
> Perhaps this will help.
> 
> https://www.youtube.com/watch?v=dhsbMmosbLU&t=75s

This approach was not known to me, but is very effective: 
1. Add tiny object. 
2. Place muliple instances
3. Rightclick, add settings, brim , select all 3 options. 
4. set options (if needed)
5. slice!

choffee, 2026-01-09:
I have been able to get something similar with the "helper_disc" from the shapes. Add it to the model and then set a modifier for that object and change the number of parameters to be different from the main model. This makes the slicer print it without changing the main model slicing.
This seems to work but it would be good to have a "mouse ears" option for brim like orca slicer does so you don't have to add them all manually.

choffee, 2026-01-09:
Looks like this is a duplicate of #2954

closed 2026-09-29 by prusaslicer-bot[bot]