prusa3d/PrusaSlicer #12571: Combine Infill Every # Layers - Not Working

CapnCox, 2024-04-10:
### Description of the bug

After just updating to 2.7.4, i'm going to print like normal but bumped up the "Combine Infill Every" to 3 layers. 
I've done this before and have clearly seen the gaps in the infill, but after slicing it no longer shows the gaps in between the infill layers. 

### Project file & How to reproduce

[Tool Kit Stand (Filament Friday) - 5783467.zip](https://github.com/prusa3d/PrusaSlicer/files/14933537/Tool.Kit.Stand.Filament.Friday.-.5783467.zip)

Load this in Prusa Slicer 2.7.4. Change the "Combine Infill Every" to 2+ layers. 
The infill does not combine every other layer like it usually does

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Windows 11

### Printer model

Prusa MK4

neophyl, 2024-04-10:
That isnt a project file.  So we dont know what settings you are using.  You need to attach a zipped up prusaslicer generated 3mf project file.  With that the devs should be able to reproduce the bug hopefully.

I just loaded the stl file you provided into PS 2.7.4 on windows 10 and  selected a default mk3 printer with a default 0.1 quality profile and changed the combine infill setting to 2 layers just as you describe and it is combining the infill properly so the layer height is double that of the perimeters.
[image]

So on the face of it it looks like there is no problem.  I do believe you are experiencing an issue but without YOUR project to reproduce it there is no way to figure out what the problem may be.

Jan-Soustruznik, 2024-04-11:
Hello, @CapnCox , could you share your 3mf(project) file, please? Thank you

closed 2026-09-29 by prusaslicer-bot[bot]