prusa3d/PrusaSlicer #12579: First layer on top of infill broken

tkircher, 2024-04-12:
### Description of the bug

The solid layer directly on top of the infill in this object is generated in two separate sections with what looks like random edges. This is causing print issues on my MK4. The layers directly above this are solid as expected.
[image]

### Project file & How to reproduce

[Untitled.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14953967/Untitled.3mf.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.2+MacOS-arm64

### Operating system

MacOS Monterey

### Printer model

MK4

tkircher, 2024-04-12:
Every kind of infill except rectilinear breaks this way, probably due to structures on top of the layers. But it makes no sense for the slicer to do this unless there's enough of a gap to join the bottom infill to the interior infill of an object above. In this case that doesn't happen.

u89djt, 2024-04-16:
@tkircher if you type "-12.9" into the Z rotation (relative) box, it'll slice with a continuous surface. It behaves differently at every location and orientation as the infill material appears different underneath it. You can see that in your example 3mf file, the slice fills the region within the smaller rectangle first by reaching the wall and extending out to the first available infill beneath it.
[image]
That's looking up through the bridge layer and the top of the infill with the wall's first layer subtracted in photoshop.
The mesh of your model has some features that might be making it more unlikely that the slicer can notice there's a single layer there. The top of that layer extends through the base of the fillet. There are also a number of places where things are 0.001mm different to what you might expect, but maybe they don't matter for slicing.
[image]
I rebuilt the model leaving in the 0.001 offsets but sorting out the surface interactions, and it generally slices better I think. If the licensing is appropriate, I'm happy to drop that rebuild here, but perhaps that's only appropriate use of this hub if the devs want a reference model without the glitches to compare.
Rectilinear infill behaves essentially the same way over that layer, but its infill spacing is about half of everything else, so reaching out to material underneath doesn't spray out beyond the wall. If you widen the wall of the object and use the same cubic infill you have, you can that it stays under the wall.
[image]

tkircher, 2024-04-19:
I did just make the walls thicker, even before this post. I'm hoping that someone can still explain why this happens and whether it's a bug or a feature. ;)

u89djt, 2024-04-19:
A subtle wrinkle that doesn't make a difference, and then a successful workaround.
I thought I'd found a specific problem, but I'm wrong. I took the surface of your mesh as a reference and it turns out to be inifinitesimally off-parallel. The difference is so small that I can't persuade Fusion to show the difference explicitly. It only shows up in the fact that a measurement from one of the horizontal faces to the XY plane shows 0.00 degrees, but no distance.
[image]
I've reoriented it so the distances show,
[image]
but it still slices the same way. Again it finds a continuous face at no Z rotation on the print bed, but 1 degree away it splits the layer. You can imagine if the surface were off-level, it would be a good idea to split the layer to be sure of being able to hit the other edges well.
And the workaround: add a small tunnel between the two regions that you want to join. Here's a project file with a tunnel added to your original mesh that slices well. The tunnel's in a randomly chosen spot and I don't know how wide it has to be, or even if it's a reliable solution in general.
[tunnel workaround.zip](https://github.com/prusa3d/PrusaSlicer/files/15041921/tunnel.workaround.zip)
[image]
[image]
Perhaps the devs will look at all of this and just think "yup, that's the way the cookie/layer crumbles". :)

closed 2026-09-29 by prusaslicer-bot[bot]