prusa3d/PrusaSlicer #12581: Extruder flowrate for first layer reverts to 0.4mm nozzle during slicing when using 0.6mm nozzle settings

jdlawley, 2024-04-12:
### Description of the bug

Consistently occurring with each version of PrusaSlicer since at least 2.6.0. Output is for Sovol SV06 (Marlin).
While slicing for a 0.6mm nozzle, the first layer extruder feedrate is generated as if the nozzle is 0.4mm (confirmed many times by achieving correct first layer flow when overriding the flowrate to 150% at the printer controller level. This only affects the first layer. I have tried setting first layer extrusion width using both millimeter and percentage values, to no avail. Material selection makes no difference. The issue does not appear in the preview window.

### Project file & How to reproduce

[Slide_Gate_Project_with_1.2mm_and_0.6mm_nozzle_gcodes.zip](https://github.com/prusa3d/PrusaSlicer/files/14964534/Slide_Gate_Project_with_1.2mm_and_0.6mm_nozzle_gcodes.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Current: 2.7.4. As noted above, been happening since at least 2.6.0

### Operating system

Windows 11 Home | 23H2 | 22631.3447

### Printer model

Sovol SV06

closed 2024-04-23 by rtyr

jdlawley, 2024-04-23:
My Z-offset does not change when printing 0.6mm vs 0.4mm. My first layer is printed at 0.2mm as a standard. Oddly, the problem does not seem to occur with 1.2mm nozzle (I only have 0.4, 0.6, and 1.2 to test).

I guess that I will have to write a script to increase the E output on the 1st layer for my 0.6mm nozzle. If the cause is not PrusaSlicer, then that implies that the problem is at the firmware level, which seems extremely unlikely to me. I'll also look at the start gcode to make sure that isn't somehow modifying the flowrate multiplier in the firmware for just the first layer. I have to re-compile firmware anyways to implement the BTT Smart Filament Sensor v2.0.

rtyr, 2024-04-23:
> My Z-offset does not change when printing 0.6mm vs 0.4mm.

Maybe that may be the issue, 0.6 may require different offset (I mean z-offset, not layer height).

jdlawley, 2024-04-23:
Total brain fart. My layer height remains the same: 0.2mm. I do adjust the Z-offset in firmware whenever I change nozzles. I'm reasonably confident that is _not_ the issue, but I will add that to the troubleshooting list next time I switch to 0.6mm.