prusa3d/PrusaSlicer #12589: Prusa Slicer crashes when slicing (MacOS14)

k1n6b0b, 2024-04-14:
### Description of the bug

Crashes with 'vector' error or segfalt and exits when slicing

> Prusa Slicer encountered an error: vector

```
[2024-04-14 06:47:55.230139] [0x000000029ef97000] [trace]   Generating perimeters for layer 9, region 0
zsh: segmentation fault  ./PrusaSlicer.app/Contents/MacOS/PrusaSlicer --loglevel=9
```

### Project file & How to reproduce

I had the models orientated on the 'short end' which is how the STL came from printables; of course would print poorly. I had been working on them in this orientation and just I rotated them face-down to re-slice. After rotating, the slicer action causes a crash every time.

Project located here: [case-rear-extended-ravemindfully-crashing.3mf](https://nextcloud.cjhx.net/s/yy5sZR4HEZYLzk6)
Error log: [case-rear-ravemindfully-crashing.rtf](https://nextcloud.cjhx.net/s/3NT4eZ42Nt5Q957)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4+MacOS-arm64

### Operating system

macOS 14

### Printer model

svol06

neophyl, 2024-04-14:
Your 2 files don't seem to be accessible.  I'm getting a blocked message when clicking on either of the links.  
If you want to attach them here you should zip them up first.  That way they can be attached directly here without putting them elsewhere.

k1n6b0b, 2024-04-15:
> Your 2 files don't seem to be accessible. I'm getting a blocked message when clicking on either of the links. If you want to attach them here you should zip them up first. That way they can be attached directly here without putting them elsewhere.

Oh whoops!  Here it is [h1-case-for-heltec-v3-running-meshtastic-model_files-crashing](https://drive.google.com/file/d/1HJr7J2St6g_B6cdPp74oqy-W57FgLGvP/view?usp=sharing)

k1n6b0b, 2024-04-16:
FYI I deleted, reimported and reorientated that one STL and it slices fine now. I expect it is the ~1* odd angle.  I'm leaving this open tho in hopes someone feels it worth while to enhance the error handling to save the next person some time!  Thanks for the great software internets!

closed 2026-09-29 by prusaslicer-bot[bot]