prusa3d/PrusaSlicer #12590: Loaded suppport enforcer not generated

steelydev, 2024-04-14:
### Description of the bug

Loading a support enforcer .stl does not always generate support when 'For support enforcers only' is selected.  In my example, switching to 'Support on build plate only' does generate the correct support. 

### Project file & How to reproduce

[FanRemoteHolder.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/14970693/FanRemoteHolder.3mf.zip)

In this file there are two support enforcers.  From the right hand pane in Slicer 2.7.4 selecting 'For support enforcers only' does not generate support in both enforcers.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.4+MacOS-x64

### Operating system

macOS 12.7.4

### Printer model

MK4

neophyl, 2024-04-14:
I don't think you understand what a support enforcer does or how it works.  You can't just put them in an area below where you want supports.  An enforcer defines areas on the Object where you want the support.  So whether that is painted or a modifier they have to define surfaces on the model that you think want support.

The reason your second one doesn't generate support is that it is not actually touching the model at all.  Your first one clearly intersects the Object and so the surfaces are defined for the support to be generated on (obeying the other settings in the support tab).  Number 2 is below the surface so not defining anything.  If you move it up very slightly you will get support.
[image]
gives this
[image]

PaparazziN, 2024-04-14:
Hi, I believe what @neophyl said is correct, so I would say it would be reasonable to close this ticket.

A tip for using support enforcers in the future is rather to use the 'Paint-on supports' function (quickly accessed by selecting the model and pressing 'L'). Using the smart fill function even allows for simply selecting a face and marking it as needing support, which is a bit faster than placing boxes in the slicer.

steelydev, 2024-04-14:
You are correct.  I didn’t understand how they work.  Sorry for the Issue!

-Geoff

===================
35 Hawthorne Pt
Little Deer Isle, ME 04650
(207) 702-1713

steelydev, 2024-04-14:
I agree, as I tried paint on and it worked great.

-Geoff

===================
35 Hawthorne Pt
Little Deer Isle, ME 04650
(207) 702-1713

closed 2026-09-29 by prusaslicer-bot[bot]