prusa3d/PrusaSlicer #12597: ARRANGE function does not work on an old 3mf file

Mdz-3D, 2024-04-15:
### Description of the bug

Hello people. I have installed update 2.7.4 and I have/have had the following problems:

- When I tried to open a file  ***.3mf** from Windows Explorer, it did it in an older version (2.0), generating version conflicts. It was not automatically updated to open with the most recent version. I solved it by going to PRUSA SLICER 2.7.4 preferences and checking the option to always open 3mf files with it. SOLVED

- I try to open a 3mf file that worked without problems in the previous version and now it gives me conflicts: It opens the file but the ARRANGE option (A) does not work correctly... it leaves the objects out of the bed even though there is enough space and in the previous version it opened 2 duplicate objects already in the center of the bed, which was where they were when the file was saved.
On the other hand, if I want to move one of the objects, it moves the 3 that are in the file: The one I want to move, the duplicate and a separate one that is also not visible.

[image]

### Project file & How to reproduce

File:
[Geoplano 10x10.zip](https://github.com/prusa3d/PrusaSlicer/files/14979803/Geoplano.10x10.zip)

Screenshots
[Grabacion de Error Prusa.zip](https://github.com/prusa3d/PrusaSlicer/files/14980004/Grabacion.de.Error.Prusa.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4+win64

### Operating system

Windows 10

### Printer model

Creality Cr6se with 0.6 nozzle

Mdz-3D, 2024-04-15:
I already have the arrange problem with differente 3mf files... one created 3 days ago with the 2.7.3 version...

shadycuz, 2024-05-27:
I dont think its just 3mf files. If I mark one item as hidden and click arrange, than I also get the bug. 

[image]

I'm not using 3mf, I loaded my files from printable by clicking the prusa slicer icon.

shadycuz, 2024-07-03:
I still have this on 2.8.0 still =/ 

@Mdz-3D Perhaps you should rename this to something like "Arrange button no longer works"

closed 2026-09-29 by prusaslicer-bot[bot]