prusa3d/PrusaSlicer #12598: Rogue layer of alternative colour in MMU multi-colour print

GuyH77, 2024-04-15:
### Description of the bug

I have a peculiar problem which see a rogue layer of a different filament in text. This has not changed in each new release hence raising it as a bug in the hope it can be looked at. In the example below you can see how the border and motif colour are printed for one single layer within the name text. I have applied some modifiers to the items in question but nothing that should cause this. I would be happy to provide the developers a sample file but I do not want to upload the files here.

[image]

[image]

[image]

It's a very strange issue but one I'd love to resolve.

### Project file & How to reproduce

Create a 3 colour print with text.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.3

### Operating system

Windows 11

### Printer model

Prusa Mk3s with MMU

cupajoe24, 2024-04-25:
I also have a very similar problem. I have three different pieces and on one piece when the surrounding top layer starts one of the pieces (but not all three for some reason?) gets a rogue layer.
[image]
[image]
[slicing_problem_tiles.zip](https://github.com/prusa3d/PrusaSlicer/files/15122325/slicing_problem_tiles.zip)

GuyH77, 2024-04-25:
Looks to be the same issue to me. Any difference in settings between the parts? Number of layer etc?

cupajoe24, 2024-04-25:
Nope, they have the same settings, the heart is cut into each tile at 0.4mm so it's the same height, it should be the same across all three but it's not.

closed 2026-09-29 by prusaslicer-bot[bot]