prusa3d/PrusaSlicer #12602: Support combine layers feature

jzagaja, 2024-04-16:
Infill has combine layers and same should be for supports.
So we print perimeters 0.1mm layer height but rest much larger.

neophyl, 2024-04-16:
Support generation already does variable layer height automatically and is not tied to your objects layer height (unless you have synchronise with object enabled).

jzagaja, 2024-04-16:
does not work on organic, prints 0,1mm

neophyl, 2024-04-16:
You didn't specify anything about support type in your original post.  

At the moment **any** sort of variable layer height does not work with Organic support types, even on the object it is supporting.  There are many issues around that already.
For example https://github.com/prusa3d/PrusaSlicer/issues/9462  (I think that is the current main one for that)

jzagaja, 2024-04-16:
That's a big drawback. Organic consumes much less filament and much easier to remove.

TerPuc, 2024-09-17:
Hello @jzagaja, thanks for your post. We already have this reported here: #9462 Therefore I close this issue as duplicate.

closed 2024-09-17 by TerPuc