prusa3d/PrusaSlicer #12606: A leaky wall

blithe74, 2024-04-17:
### Description of the bug

When using the Classic engine, walls are generated with many holes (see screenshot).
[image]

### Project file & How to reproduce

1) Set parameters how in Classic.gcode file from archive.
2) Set engine "Classic".
3) Generate a Gcode.
See a problem.
[Model.zip](https://github.com/prusa3d/PrusaSlicer/files/15015475/Model.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4+win64

### Operating system

Windows 10 Pro 22H2

### Printer model

Merlin compatible

neophyl, 2024-04-17:
First of all to debug an issue the devs need a Prusa Slicer project file, not just a model file.  The settings you are slicing with are just as , if not more, important.

For example if I load your model into a Mk3 0.2 quality profile and slice it slices fine with zero holes.  Even with the classic slicing option selected.
[image]

To know for sure the devs need your project.   Just slice so you get the issue then use File>save project as to save a PS .3mf file.  The take that 3mf file and ZIP it up and attach it here like you did with the model file.

Usually issues like this are caused by the wall thickness of the model being thinner in places than your configured extrusion width.  Any curve in the stl file is made up of a series of triangles.  This leads to varying wall thickness, even if only fractionally.  You may think your wall is the same thickness all along but it is not.  If your nominal wall closely matches your extrusion width then some parts can fall below that extrusion width and so the slicer will omit them.  
Arachne is designed to vary the extrusion width as kit goes along.

If you have to use classic then common remedies is to 
a- configure a smaller extrusion width
b - turn on detect thin walls (which can introduce its own issues)
c- apply a XY compensation in effect making all your model wall thicknesses larger
d - edit the model to have walls that are thicker and exceed the extrusion width you are using.

blithe74, 2024-04-18:
С уважением, Александр Шуковилов

closed 2026-09-29 by prusaslicer-bot[bot]