prusa3d/PrusaSlicer #12607: Custom G-code is removed (cleared) when new model is added or model is replaced

dmak, 2024-04-18:
**Steps to reproduce the issue**

* Add custom G-code at a give layer by right-click on `(+)` sign on layers' ruler.
* Add new model to the project or replace the current model from context menu.

Result: custom G-code is removed.

Expected: custom G-code is preserved and/or the user is asked "Would you like to keep custom G-code?"

I takes some time to place a needed G-code at specific layer, e.g. turn on the fan. Very pity I had to do that each time I replace the model with a new one, which has only slight changes.

Additionally: Context menu reads "Replace with STL" however allows to replace with 3MF/OBJ/etc, so it's better to rename it to "Replace model"

[Initially posted to forum](https://forum.prusa3d.com/forum/prusaslicer/custom-g-code-is-removed-cleared-when-new-model-is-added-or-model-is-replaced/).

Jan-Soustruznik, 2024-04-22:
Hello, @dmak , 

Custom G-codes are deleted when the model is changed or the scene is completly deleted - that is expected behaviour. Anyway If you need to add more globaly some custom g-code example "turn fan at specific layer height" you can use macros, go to `Printer Settings -> Custom G-code -> block "Before layer change G-code"`, and use this example `{if layer_z == 5}; example G-code {endif}`(this macro add at layer height 5mm ";Sample G-code"). more about the macros, link: https://help.prusa3d.com/article/macros_1775

dmak, 2024-04-22:
I still think that is impractical to remove custom G-code when the model is reloaded, as in my case these are small changes to the model. But the workaround with macro which you suggested is good enough for my purpose. Thanks for suggestion!

closed 2024-04-22 by dmak