prusa3d/PrusaSlicer #12608: App Crashed while generating G-Code

miko671995, 2024-04-18:
### Description of the bug

Hello. Im actally fighting agains the problem that prusaslicer suddenly making my windows crashing. 

### Project file & How to reproduce

My observations:
Large Files are doing this issue.
It happens mostly in 90 of process slicing (generating G-code) 
Screen will stay white with windows circle loading and app crashed.  After i open task manager to forcely close it windows CRASHED.  
report from windows crash file:
`Version=1
EventType=AppTermFailureEvent
EventTime=133577358234779578
ReportType=4
Consent=1
ReportStatus=2049
ReportIdentifier=a6e44bae-fa6d-4a4c-8fe3-4208ff201c3f
IntegratorReportIdentifier=8ddb8b4d-adce-4dfa-8eee-668a175649d3
Wow64Host=34404
NsAppName=AppTermFailureEvent
OriginalFilename=prusa-slicer.exe
AppSessionGuid=000025dc-0001-01d7-9bd5-cfc2e58fda01
TargetAppId=W:0000ab86432f2dd34409693205b8f490b55c0000ffff!0000fe32b5e9fe9462ebd7cf5c8917d675963e8cec86!prusa-slicer.exe
TargetAppVer=2024//04//05:09:32:43!3f8d3!prusa-slicer.exe
BootId=4294967295
TargetAsId=634
UserImpactVector=269484336
IsFatal=1
EtwNonCollectReason=1
Response.type=4
DynamicSig[1].Name=Verzia OS
DynamicSig[1].Value=10.0.19045.2.0.0.768.101
DynamicSig[2].Name=Identifikácia miestneho nastavenia
DynamicSig[2].Value=1051
OsInfo[0].Key=vermaj
OsInfo[0].Value=10
OsInfo[1].Key=vermin
OsInfo[1].Value=0
OsInfo[2].Key=verbld
OsInfo[2].Value=19045
OsInfo[3].Key=ubr
OsInfo[3].Value=4291
OsInfo[4].Key=versp
OsInfo[4].Value=0
OsInfo[5].Key=arch
OsInfo[5].Value=9
OsInfo[6].Key=lcid
OsInfo[6].Value=1051
OsInfo[7].Key=geoid
OsInfo[7].Value=143
OsInfo[8].Key=sku
OsInfo[8].Value=101
OsInfo[9].Key=domain
OsInfo[9].Value=0
OsInfo[10].Key=prodsuite
OsInfo[10].Value=768
OsInfo[11].Key=ntprodtype
OsInfo[11].Value=1
OsInfo[12].Key=platid
OsInfo[12].Value=10
OsInfo[13].Key=sr
OsInfo[13].Value=0
OsInfo[14].Key=tmsi
OsInfo[14].Value=222626075
OsInfo[15].Key=osinsty
OsInfo[15].Value=1
OsInfo[16].Key=iever
OsInfo[16].Value=11.3636.19041.0-11.0.1000
OsInfo[17].Key=portos
OsInfo[17].Value=0
OsInfo[18].Key=ram
OsInfo[18].Value=24457
OsInfo[19].Key=svolsz
OsInfo[19].Value=222
OsInfo[20].Key=wimbt
OsInfo[20].Value=0
OsInfo[21].Key=blddt
OsInfo[21].Value=191206
OsInfo[22].Key=bldtm
OsInfo[22].Value=1406
OsInfo[23].Key=bldbrch
OsInfo[23].Value=vb_release
OsInfo[24].Key=bldchk
OsInfo[24].Value=0
OsInfo[25].Key=wpvermaj
OsInfo[25].Value=0
OsInfo[26].Key=wpvermin
OsInfo[26].Value=0
OsInfo[27].Key=wpbuildmaj
OsInfo[27].Value=0
OsInfo[28].Key=wpbuildmin
OsInfo[28].Value=0
OsInfo[29].Key=osver
OsInfo[29].Value=10.0.19041.4291.amd64fre.vb_release.191206-1406
OsInfo[30].Key=buildflightid
OsInfo[30].Value=d847bcf7-4007-40be-b425-c99268d4dbd2
OsInfo[31].Key=edition
OsInfo[31].Value=Core
OsInfo[32].Key=ring
OsInfo[32].Value=Retail
OsInfo[33].Key=expid
OsInfo[34].Key=fconid
OsInfo[35].Key=containerid
OsInfo[36].Key=containertype
OsInfo[37].Key=edu
OsInfo[37].Value=0
File[0].CabName=prusa-slicer.exe.atk.kdmp
File[0].Path=WER18CB.tmpatk.kdmp
File[0].Flags=589824
File[0].Type=5
File[0].Original.Path=\\?\C:\Windows\Temp\WER18CB.tmpatk.kdmp
File[1].CabName=AppCompat.txt
File[1].Path=WER18CC.tmp.appcompat.txt
File[1].Flags=17367042
File[1].Type=5
File[1].Original.Path=\\?\C:\Windows\Temp\WER18CC.tmp.appcompat.txt
File[2].CabName=WERInternalMetadata.xml
File[2].Path=WER18ED.tmp.WERInternalMetadata.xml
File[2].Flags=851970
File[2].Type=5
File[2].Original.Path=\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER18ED.tmp.WERInternalMetadata.xml
File[3].CabName=WERInternalRequest.xml
File[3].Path=WER18FD.tmp.xml
File[3].Flags=851970
File[3].Type=5
File[3].Original.Path=\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER18FD.tmp.xml
File[4].CabName=memory.csv
File[4].Path=WER190C.tmp.csv
File[4].Flags=589826
File[4].Type=5
File[4].Original.Path=\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER190C.tmp.csv
File[5].CabName=sysinfo.txt
File[5].Path=WER192C.tmp.txt
File[5].Flags=589826
File[5].Type=5
File[5].Original.Path=\\?\C:\ProgramData\Microsoft\Windows\WER\Temp\WER192C.tmp.txt
File[6].CabName=Report.zip
File[6].Path=Report.zip
File[6].Flags=65536
File[6].Type=11
File[6].Original.Path=\\?\C:\WINDOWS\system32\Report.zip
Ns[0].Name=stopcode
Ns[0].Value=CFFFFFFF
Ns[1].Name=p1
Ns[1].Value=1
Ns[2].Name=p2
Ns[2].Value=0
Ns[3].Name=p3
Ns[3].Value=0
Ns[4].Name=p4
Ns[4].Value=0
FriendlyEventName=Stopped responding and was closed
ConsentKey=AppTermFailureEvent
AppName=PrusaSlicer
AppPath=C:\Program Files\Prusa3D\PrusaSlicer\prusa-slicer.exe
ReportDescription=A problem caused this program to stop interacting with Windows.
NsPartner=windows
NsGroup=windows8
ApplicationIdentity=00000000000000000000000000000000
MetadataHash=-291857606`

Im not able to do a screenshot because after it happends my windows will freeze and i have to reboot my PC.
[Prusareport.zip](https://github.com/prusa3d/PrusaSlicer/files/15024477/Prusareport.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Version 2.7.4+win64

### Operating system

Windows 10 Home 22H2

### Printer model

I3 MK3.5+

Jan-Soustruznik, 2024-04-19:
Hello, @miko671995 , Thank you for your report, but we cannot reproduce crash with your attached 3MF file. Could you share some more info, example: 1. HW spec of your PC? 2. it starts in 2.7.4? 3. what apps runs with PS on PC together during crash? 4. did you have open more instances of the PS? Thank you, have a nice day

miko671995, 2024-04-20:
Hello. The only thing what is happening as far as i catched is that its doing from version 2.7.2. I was printing lithophanes with bigger resolution so files was big enough. It is doing really randomly. But good to mention is that First time that im slicing i sucessfully get Gcode but if i wanted to change any parameter like the infill or what ever on second slicing app crashed (whole windows)  
[DxDiag.txt](https://github.com/prusa3d/PrusaSlicer/files/15047479/DxDiag.txt)
there is list of my HW from DXDIAG. Maybe im thinking about issue coused by driver compatilibity. could i do something to help you debugging this behaviour. I think its doing only on my PC Clean install was not worth it.

Jan-Soustruznik, 2024-04-22:
Hello, @miko671995 , Could you check if you some some HW issue, example with HDD or RAM. Where the PC completly crash, that may be HW issue that not depend on the software or drivers.

miko671995, 2024-04-24:
I tested all of my hw with stress test. Strange behaviour is that this problem occurs only with lithophanes.

closed 2026-09-29 by prusaslicer-bot[bot]