prusa3d/PrusaSlicer #12610: Make the "layer time goal" slow down printing only infill and internal perimeters.

shinanca, 2024-04-18:
**Is your feature request related to a problem? Please describe.**
Yes and no

**Describe the solution you'd like**
Almost all printers have ripples when printing at certain speeds. Sometimes there are no ripples only at high speeds (for example 100+ or ​​150+). This makes it difficult to print small parts with short layer times. You have to print several pieces, or print “garbage” next to each other to increase the layer time.
At the moment, "Layer time goal" slows down the printing of all types of lines, it would be very convenient if it were possible to slow down only the filling, or sequentially - filling - internal perimeters, external perimeters.
Those. we print the outer perimeters at 100, and the infill at least at 5-20mm/s

**Additional context**
need max slow down % parameter for external perimeters, for other layers "Layer time goal", and slow down by priority: infill>internal perimeters>external perimeters.
Print will look more consistency.

Tupson444, 2024-04-18:
Also, when the external perimeter slows down, often the plastic looks more shiny because of slower speed, which looks bad (different shine at different layer heights). If only internal perimeters and infil are slowed down as suggested here, this problem would be solved.

Tupson444, 2026-09-18:
still relevant