prusa3d/PrusaSlicer #12611: Profile Inheritance: UI:  Provide a Button: Hide/Show Parent Entries.

Andrei-Pozolotin, 2024-04-18:
Currently, in the Version 2.7.4+linux-x64-GTK3,

When using profile inheritance for `Print Settings`, `Filament Settings`, etc, then:

Settings Entries inherited from the parent profile are shown:
* in the tree-view: as "magic green text"
* in the panel-view: as "magic green text + closed gray locks" vs "plain black text + open orange locks"

Which creates needless visual noise, if one's focus is only on the inheritance relationship review.

[image]

This issue is a request to provide in the UI a simple checkable button named `Show Settings Override`, 
which, when pressed/checked, will do the following:
* in the tree-view: hide all non-modified parent nodes
* in the panel-view: hide all non-modified parent entries

In other words, the new button-activated UI will look similar to the following:

[image]

Jan-Soustruznik, 2024-04-22:
Hello, @Andrei-Pozolotin , Thank you for you feature request, but I think that we already implemented requested feature that you mention. You can use "Compare Presets" feature, where you can compare what is different in current preset with anohter one, example the parent profile. Comparation you can find on top panel near search icon. I attached screen. [image]

Andrei-Pozolotin, 2024-04-23:
@Jan-Soustruznik: thank you for your note.

Yes, I was aware of the "Compare Presets" feature. 

It just feels like it needs 10 times more work to get the result :-)

Namely:
* first, click on "Dependencies" to find out the name of the parent, if any
* next, click on  "Compare Presets" 
* since there is no search in  "Compare Presets" combo box, 
* scroll through 1 or 2 hundred of existing presets, trying to find the right parent
* now review the difference and decide that you do want to change the parent preset
* great, now close the  "Compare Presets" dialog, which does not react to "Escape", need to hunt for "X"
* navigate to the "Print/Filament Preset" combo box
* search again through 1 or 2 hundred of existing presets, trying to find the right parent
* finally, change the parent
* no, not finally: need to go back to the original child preset 
* that is, by search again through 1 or 2 hundred of existing presets

you see: 10 more steps - and what about if you want to go 2-3 parent levels down?

which means that there is also a need for "Go Past" / "Go Next" present navigation buttons :-)

closed 2026-09-29 by prusaslicer-bot[bot]