prusa3d/PrusaSlicer #12613: Add a Zig Zag infill that works like the one in Cura (Line infill doesn't do the job)

foormowniki, 2024-04-18:
**Is your feature request related to a problem? Please describe.**
I'm working on 3D-printed shoes like Zellerfeld. They use 0 perimeters with Zig Zag infill. Cura generates nice patterns on the surface of the object, which, at the same time, keeps the holes in between infill, and those are used as ventilation holes in the shoes. Here is an example of the outcome:
[image]

If you want to reproduce this in PrusaSlicer using line infill, you will observe a resetting of the positioning of the infill pattern along the Z axis, which ruins the pattern and desirable effect of keeping the holes unblocked. Here is an example:
[image]

**Describe the solution you'd like**
Copy a ZigZag algorithm from Cura to reproduce the surface pattern while printing without a perimeter. I think it is coded the way it makes sure every node of infill stays on top of each other, whether in prusa it is prioritizing geometry representation

**Describe how it would work**
Create an additional infill pattern called zig zag

**Describe alternatives you've considered**
Switching to Cura

**Additional context**
Shoe 3D printing is a huge and fast-growing market. Many companies are exploring this, but very few are using Prusa just because this one feature is missing.

inventabuild, 2024-06-04:
Great infill suggestion!  Is there an update on when the Zig Zag infill feature will be implemented in PrusaSlicer?

pyou2123, 2024-06-20:
Same here!! still in waiting zigzag infill in pursaslicer!!!!

pyou2123, 2024-06-20:
[image]
[image]
compare different infill between pursa and Cura zigzag

foormowniki, 2024-06-21:
@pyou2123 exactly! Thanks for the close-up pics! @josefprusa we need this so bad!:D

JaredBatch, 2024-06-26:
Looking for this as well!

fabriziodelucia, 2024-06-26:
@josefprusa yeah make it happen pls

pyou2123, 2024-07-10:
help plz!!! we need this feature updates :) @josefprusa

raresmanolache, 2024-08-01:
@josefprusa please help us!

closed 2024-09-12 by lukasmatena

VickyTGAW, 2025-05-06:
Zig Zag Infill Enthusiasts - I wanted to let you know that 3DPrintBunny recently released an "[Infill Daisy Bowl](https://makerworld.com/en/models/1360429-infill-daisy-bowl#profileId-1404945)" which lends itself well to the zero perimeters technique. In fact, she designed it for zero perimeters!

[image]

P.S. If you are looking to control the Zig Zag angles in your bowl or other projects, there is a [pull request with some code](https://github.com/prusa3d/PrusaSlicer/pull/14073) that could help.