prusa3d/PrusaSlicer #12616: Unexpected Behaviour With MM Painting - Weird Lines

ImTooks, 2024-04-19:
### Description of the bug

When using the 'Multimaterial painting' tool, the sliced model has lots of unexpected extrude lines which would run the print. See below for images:

**Painted model**
[image]

**Sliced result of painted model**
[image]

**Sliced cross section of painted model**
[image]

**Sliced result of unpainted model**
[image]

**Sliced cross section of unpainted model**
[image]

Any help would be appreciated, I will attach the 3MF for reference. Cheers!

### Project file & How to reproduce

[Pikachu_BugTest.zip](https://github.com/prusa3d/PrusaSlicer/files/15033895/Pikachu_BugTest.zip)

Steps to reproduce:
1. Slice file

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.4+win64

### Operating system

Windows 10 Home Version 10.0.19045 Build 19045

### Printer model

Voron Trident 300 w/ ERCF

Jan-Soustruznik, 2024-09-26:
Hello, @ImTooks , thank you for your feedback, we will try to fix in some next PrusaSlicer release. Have a nice day

lukasmatena, 2024-11-21:
Fixed in [2.9.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.9.0-alpha1). Closing.

closed 2024-11-21 by lukasmatena