prusa3d/PrusaSlicer #12617: MMU Single Material Mode is missing for MMU3 on MK4 printer

sch000f, 2024-04-19:
As of 2.7.4 it is not yet possible to use MMU3 on the MK4 printer in Single Material Mode and print with filaments other than PLA and PETG.

https://help.prusa3d.com/guide/11-first-flight_357309
Step 16 shows that this is already implemented for the MMU3 on MK3S/MK3S+ printers.

Without this feature the MK4 printer loses the abillity to print anything besides PLA and PETG.

rtyr, 2024-04-19:
It's not missing, it's just done differently than on older printer models (like MK3S). You should use standard MK4 (non-MMU) profiles for single prints. Printer will recognize it and will let you choose the slot for printing.

If you have further questions, please contact our tech support at www.prusa3d.com.

[image]

closed 2024-04-19 by rtyr

sch000f, 2024-04-19:
Thank you for the quick reply.
This should be mentioned in the MMU3 setup guide.

rtyr, 2024-04-19:
I notified responsible department.