prusa3d/PrusaSlicer #12620: Ui bug in select box

JaquesBoeno, 2024-04-20:
### Description of the bug

when I select things like the print profile happens this bug, the other items become increasingly clear, making it impossible to read, exemplify better in the video. the only way to read is using the arrow keys, this bug happens in version gtk3 2.7.4, but in other tests I did, all versions from 2.7 contain this same bug, only working properly in 2.6.x

### Project file & How to reproduce

[Screencast from 2024-04-19 21-56-26.webm](https://github.com/prusa3d/PrusaSlicer/assets/78875949/d6c65735-e7bd-4fb0-8d62-626f2128f6d5)

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

Prusa slicer 2.7.4

### Operating system

archlinux

### Printer model

Ender 3

ErgoPraxis, 2024-07-19:
https://github.com/prusa3d/PrusaSlicer/issues/13095#issue-2407730195

Looks like I'm not the only one 🙂

moritzschepp, 2025-01-03:
Same problem here, also on archlinux (XFCE) and PrusaSlicer 2.9.0.

I can reset the menus by restarting PrusaSlicer or by changing the "Default palette for mode markers:" in the preferences, maybe that helps ...

moritzschepp, 2025-01-07:
I also noticed that the problem isn't there in the profile selector menu under "Print Settings".

closed 2026-09-29 by prusaslicer-bot[bot]