prusa3d/PrusaSlicer #12622: Linear Advance G-Code for Prusa MINIIS

koppensb, 2024-04-20:
### Description of the bug

In the PETG or PLA filament settings for the MK3.5 the start G-Code has settings for Linear Advance (M900) with ifs for the MINIIS but in the filament settings for the MINIIS the G-Code is missing. Is this desired because Linear Advance G-Code is not necessary for the MINIIS or a bug?
[image]
[image]

### Project file & How to reproduce

File -> Filament Settings MINIIS or MK3.5

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

MacOS 14.4.1

### Printer model

MINIIS

rtyr, 2024-04-20:
It is correct/intended. IS fw uses M572 commands for pressure advance, M900 is not emitted for 3.5/MINIIS (the condition is negative).

closed 2024-04-20 by rtyr

koppensb, 2024-04-20:
Thanks for the clarification. Now I see the the error in interpretation of the if statements.